gab-k/RP2040-Decoder #11: Motor not controlled

chenhongmai, 2023-06-19:
Hi Gab-k,

I have the PCB made by JLC and soldered with components. I could verify that the function works as the lights are working corresponding to the CV.h. However, after the initial tuning the motor is running at full speed and not controlled (as long as the decoder is connected to the track) and the motor IC and bridge gets extremely hot...(very similar to the other post...)

I also tried to solder a 1210 resistor to the motor ports as a load, but probably due to previous problems with motor running at full speed it won't work (very hot...)

What is your thoughts on the problem?
Thanks for your help.

gab-k, 2023-06-19:
Try Re-running the motor calibration after you are certain that your motor works, perhaps by supplying it with a DC voltage directly.
Regarding the function mapping make sure your light works.

chenhongmai, 2023-06-26:
Hi Gab, 

Thank you for your help. 
Another question, is POM programming on main mode possible? 
Thanks.

gab-k, 2023-06-26:
Certainly possible but not currently implemented in Software.

chenhongmai, 2023-07-07:
> Try Re-running the motor calibration after you are certain that your motor works, perhaps by supplying it with a DC voltage directly. Regarding the function mapping make sure your light works.

Hi,

I made a test PCB to test all the functions, and I tried to map F0 forward to the LED and is still not working. But the exact same LED works for the other channels. I think there might be a bug somewhere in the directional light outputs.

In regards to the motor control, after tuning, the motor is still running full speed without being controlled.
Our test equipment:
ESU lokprogrammer
ESU decoder tester

We tried to re-tune the motor but it does not work, still running full speed. 
The test equipments are all good.

Thanks for helping...still try to see why this happened. Hopefully these information helps for debugging.

gab-k, 2023-07-15:
> In regards to the motor control, after tuning, the motor is still running full speed without being controlled. Our test equipment: ESU lokprogrammer ESU decoder tester

I just merged a pull request kindly provided by questor, it adds functionality for logging various parameters using uart or usb, this might help you troubleshoot. If the motor runs at full speed, this indicates that the I component of the PID controller is rising too high or that the calibration is incorrect.

> I made a test PCB to test all the functions, and I tried to map F0 forward to the LED and is still not working. But the exact same LED works for the other channels. I think there might be a bug somewhere in the directional light outputs.

Unfortunately I cannot really help you with this issue at the moment as I lack the time to do so, however I encourage you to resolve the issue on your own, as it can't be all too complicated considering it involves only toggling a GPIO pin.

closed 2023-10-28 by gab-k