gab-k/RP2040-Decoder #14: Add USB-C connector

chatelao, 2023-10-28:
I love this project and think it has the potential to shake the Model-Railroad community.

The great thing about the RP2040 is the included permanent ROM uf2 Bootloader. Yes, adding a USB-C interface will cost some realestate, BUT would lower the bar for developers signifcantly, removing the need to understand SWD programming at all.

I think, as soon as the Software compiles in a Ardunino environment, the support of entry enthusiast will explode.

gab-k, 2023-10-28:
I agree and also thought about adding the USB connector before. If I have the time I might change the PCB with an optional USB connector on the back side in the future.

chatelao, 2023-10-29:
It would be great - I tried to do it myself, but failed misarably.

I would like to try to add a Märklin (MM2) supported firmware version.

btw: Reading and Writing a config file with all CVs and Parameters over a USB-C  could greatly simplyfing the process of decoder setup. Even sound files in a microcontroller compatible format could be added "drag n drop".

closed 2024-05-05 by gab-k

chatelao, 2025-09-13:
Oh, die USB Version sieht sehr schön aus. Ich hab mir überlegt, ob man mit senkrechten Buchsen etwas weniger Platz auf der Platine bräuchte.