gab-k/RP2040-Decoder #16: nmake not found

Space-Farmer, 2023-12-12:
Hello, 

I have an issue during build process. 
I got the following error:

PS C:\Users\danie\Documents\Pico-v1.5.1\RP2040-Decoder\build> cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk')
PICO_SDK_PATH is C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
CMake Error at CMakeLists.txt:6 (project):
  Running

   'nmake' '-?'

  failed with:

   Das System kann die angegebene Datei nicht finden

-- Configuring incomplete, errors occurred!
See also "C:/Users/danie/Documents/Pico-v1.5.1/RP2040-Decoder/build/CMakeFiles/CMakeOutput.log".

Could it be the case that I have to install some Microsoft tool what is not mentioned?

gab-k, 2023-12-13:
Im not sure I personally don't have a windows machine at the moment so I can't try. Make sure you have Build tools for visual studio installed Section 9.2.2.3 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

If that does not work I would suggest following the steps described from section  9.2.2 onwards.

Space-Farmer, 2023-12-13:
Thanks for the help. 
I thought that with the PICO SDK installer, all files and application for compilation etc. are installed. 
Unfortunatly that was not the case....

Solution for me was to install the MS Build tools. (like described in the manuel from above, chapter 9.2.1)
After installation open the Developer command line for VS2022, then open the pico-env.cmd from RaspiPico SDK. 
Then it compiles perfect.

closed 2023-12-13 by gab-k