gab-k/RP2040-Decoder #18: Function and light control

Space-Farmer, 2023-12-20:
Sorry to ask the next stupid question... 
Great project that you shared!! 
Setting into operation to drive forward and backward is already working nicely.
Today I started and want to controll the LEDs.
I tested the LEDs with grabbing the Voltage behind the rectivier (+V and GND), they are working fine.
Now I added each LED Row white/red color ; front/backside to one of the AUX pins.
[image]
I experimented with the CVs 257,261,265, 269 but it did not change anything.
I came to the point maybe I have to enable the PWM to the pins that they are modifyable.
[image]
For the PWM I used the values from wiki:
[image]
Which results to for GPIO 24,25,26,27
[image]
Additionaly I doublechecked the CV for Fx forward/reverse:
[image]
Please find also attached a readout of all CVs

[CVs.csv](https://github.com/gab-k/RP2040-Decoder/files/13733560/CVs.csv)

Thanks.

gab-k, 2023-12-21:
Hi, I dont really clearly understand what you are trying to achieve? You are saying the LEDs are always on. When should the LEDs be off? Depending on direction?

Space-Farmer, 2023-12-24:
Hi, sorry, was a little late that day.

What I wanted to achieve was:
That the AUX0 is front light (forward direction) and only on when moving forward & Function0 enabled.
AUX1 is front light (reverse direction) and only on when moving reverse & Function0 enabled.
AUX2 is rear light (forward direction) and should only on when moving forward and pressing Function1 button.
AUX3 is rear light (reverse direction) and should only on when moving reverse and pressing Function1 button.

What happens: all lights are always on. No change with direction nor with enabling the Functions buttons inside visualization.

Did I miss something to enable for the application described above?
Is PWM and function control possible in parallel?
Did I need to consider the Signal negation by N-Mos of GPIO24-27?

Many thanks for your work on that project.
I wish you a merry Christmas and a happy new year.

Best Regards.

gab-k, 2023-12-26:
> Did I miss something to enable for the application described above?

Your CV's are mostly correct, bits 7 to 4 should not be set in CV257, CV261, CV265 and CV269. 
So it should be CV257 = 1, CV261 = 2, CV265 = 4 and CV269 = 8 also I made a mistake in my description saying that the clock divider should be configured with a value of 10 to achieve 10kHz. In reality it needs to be set to a value of 9 in order to achieve a clock divider of 10 as a value of 0 corresponds to a divider of 1. However it should not cause the issue you have described as it only slightly alters the pwm frequency and has nothing to do with the issue you are describing.

> Is PWM and function control possible in parallel? Did I need to consider the Signal negation by N-Mos of GPIO24-27?

There isn't any negation to consider because when the gpio connected to the mosfet gate is set to 3.3V / "1" the mosfet will conduct and therefore connect the load to GND, closing the circuit and lighting the light. Regarding the first question, yes functions can be configured as PWM or regular output.

I tried to reproduce your issue but was not able to, you should draw a sketch of how you wired your lights (maybe on paper doesnt have to be anything special) to eliminate the possibility of wrong wiring. Also please try to set the CVs the way i mentioned above.

Space-Farmer, 2023-12-27:
>  bits 7 to 4 should not be set in CV257, CV261, CV265 and CV269
Reason why I marked the bit as "0" not as "1" was that all unused CVs where filled 255. And to distinguish between not configured and configured with "1" to "0" made more sense to me in that moment. Sorry that I did not had a deeper look to the code. 

> So it should be CV257 = 1, CV261 = 2, CV265 = 4 and CV269 = 8
I made the changes also CV258,259,260,262,263,264,266,267,268,270,271,272 =0 to ensure that there is no interference. All LEDs are still always on. Maybe I missed some other CV, I will have a deeper look.
(when I change to program mode, LED are off, seems no short circuit) 

> There isn't any negation to consider because when the gpio connected to the mosfet gate is set to 3.3V / "1" the mosfet will conduct and therefore connect the load to GND
That fits to my thoughts, thanks. 

> draw a sketch of how you wired your lights
Let me try this in text form.
PCB V+ -> LED Anode
LED cathode -> resistor
resistor -> Pin header male
Pinheader female -> PCB AUX

SMD LEDs in parallel, 2 red 3 white. TOP-LED Case 3528, continues forward current: 50mA
red resistor 2kR, white resistor 500R 

My next steps:
factory reset of the decoder
check with other decoder PCBs

gab-k, 2023-12-27:
> Reason why I marked the bit as "0" not as "1" was that all unused CVs where filled 255. And to distinguish between not configured and configured with "1" to "0" made more sense to me in that moment. [...]
It seems odd to me that your CVs are not the default value (see CV.h) normally the decoder should save the default values in  persistent flash memory after first flashing the software.

> I made the changes also CV258,259,260,262,263,264,266,267,268,270,271,272 =0 to ensure that there is no interference. All LEDs are still always on. Maybe I missed some other CV, I will have a deeper [...]

Bits 7 to 4 should not do anything anyway as the bits are always logical & masked with a bitmask containing the allowed pins only.

> Let me try this in text form. PCB V+ -> LED Anode LED cathode -> resistor resistor -> Pin header male Pinheader female -> PCB AUX
>
> SMD LEDs in parallel, 2 red 3 white. TOP-LED Case 3528, continues [...]
Looks good as far as I can tell.

> My next steps: factory reset of the decoder check with other decoder PCBs

Resetting would have been my next suggestion as well.

Space-Farmer, 2024-01-02:
I flashed the device with an spi programmer not with picoprobe.
Picoprobe is not running well on my side and with the spi programmer and a contact clamp it is very convinient to programm more parts in a row. (to get the bin files you have to uncomment one line inside cmake dile) 

After resetting the default values where in Place.

And yes light and functions are working fine. Thanks for the Support.
Issue could be closed from my side. If you have further quesitions or you need readouts from the spi flashed parts feel free to ask.

closed 2024-01-02 by gab-k