gab-k/RP2040-Decoder #24: Marklin

alsig, 2024-02-03:
Hi,

I have just received a batch of decoders from JLCPCB. I'm able to communicate with the boards using the SWD interface and have successfully programmed the first.

I experiencing different issues. I have fried a couple of motor drivers.  I suspect that since I'm running Marklin the voltage levels are somewhat different than a DC/DCC system. The motor driver is rated for maximum of 26V but I can read 19V from my CS3. But that's properly an AC of some sort... 

Another issue that I have is one board where the U1, the voltage regulator, get extremely hot after a few seconds. I'm able to connect to the mcu through SWD before I shut the power off... So something is working but other parts are drawing too much power...

So in short, have any one successfully tried these decoders on Marklin tracks? 

Or do anybody have an idea to how I can modify the PCBs so they can work with Marklin?

/Jens

gab-k, 2024-02-09:
The issue regarding the motor driver seems odd, can't imagine that the CS3 puts out voltages > 26V. Regarding the voltage regulator, yes it is a bad design, it gets very hot especially at higher track voltages due to its linear regulation nature. An ugly but easy fix would be to just solder another regulator in something like a SOT-223 package (or similar) on the back side of the board.

probonopd, 2024-02-13:
Wouldn't some "rectified, step-down switch-mode converter with built-in internal power MOSFETs" ([example](https://www.monolithicpower.com/en/products/power-management/power-modules/mpm3612-33.html); I am not necessarily suggesting this particular model) be able to do the job more nicely than a linear (=inefficient and hot) converter?

gab-k, 2024-02-14:
In terms of efficiency its certainly the better option. However i decided against it due to higher cost and increased need for PCB space. Additionally the availability / potential replacements with the same pinout are better with an LDO. Replacing it with an SOT-89 part however might be a solution.

closed 2024-05-05 by gab-k