gab-k/RP2040-Decoder #33: Documentation quiestions

voroskoi, 2025-01-19:
Hi,

First of all, thank You for this work and sharing them online!

After reading the documentation I have a couple of quiestions. I was hoping You could answer them and I could make a PR with the clarifications for You in return.

1. "You can assemble the board yourself, but it’s best not to do so if you’re not familiar with soldering small components. Alternatively, you can order the board fully or partly assembled." - I have never ordered any PCB. I assume You talk about the possibility of ordering the SMD parts and soldering those together. On JLCPCB site there is an option for assembling the board, which everyone should take if they want to provide BOM and Pick and Place files and want a fully working board after the purchase, but They can also decide to order the board and solder everything at home. Is that right?
2. "Soldering the back of the PCB by hand is not too difficult." - On JLCPCB there is an option for "Assembly Side" with top, bottom and both sides. If I chose top side (which is the default) that means I have to make some soldering at home? Or You just happen to mention that the bottom half soldering is not that hard if somebody is already doing the top part?
3. I have not been able to figure out what to do with `netlist.ipc` and `designators.csv` while ordering the PCB. Do I need those?
4. On the [wiring](https://gab-k.github.io/RP2040-Decoder/sphinx/html/hardware/rev_1_0/hw_description.html#wiring-the-decoder) diagram: The `Load1` is using 3.3V to switch a LED or something like that. `Load0` on the other hand gets the DCC signal and voltage (up to 24V) directly when switched on, right? Can You name a use case for that?

Thank You!

gab-k, 2025-01-19:
> I have never ordered any PCB. I assume You talk about the possibility of ordering the SMD parts and soldering those together. On JLCPCB site there is an option for assembling the board, which [...]

Yes, that is correct. 

>  If I chose top side (which is the default) that means I have to make some soldering at home? 

Yes, you would have to order the parts for the bottom side and assemble the bottom side by hand.

> Or You just happen to mention that the bottom half soldering is not that hard if somebody is already doing the top part?

Yes, that is basically what I meant, assembly of the top side and doing the bottom side yourself is a little bit more price effective. Assembling everything yourself is of course even cheaper.

>     3. I have not been able to figure out what to do with `netlist.ipc` and `designators.csv` while ordering the PCB. Do I need those?

No, you don't need those.

> On the [wiring](https://gab-k.github.io/RP2040-Decoder/sphinx/html/hardware/rev_1_0/hw_description.html#wiring-the-decoder) diagram: The `Load1` is using 3.3V to switch a LED or something like that. [...]

Mostly correct, the load gets connected to VDC which is the rectified DCC voltage. The transistor then basically acts like a switch and switches the other side to GND potential.

> Can You name a use case for that?

You might have a classic 12V or 24V Halogen/Incandescent light for lighting.

voroskoi, 2025-01-21:
Thank You very much for those answers.

I have tried to use PCBWay as they have an order sharing system which makes ordering much easier for noobs like me, but they replied with a question I do not really understand:

> Hello Dear,
> Thanks for your inquiry on PCBWay.
> As shown in attached screenshot.
> There are some vias be designed on the SMD pads, pls advise if proceed with via-in-pad process for all of the via holes?
> If yes, this service is chargeable (over $100usd) and vias are to be non-conductive epoxy filled, plated over with copper and planarized.
> If not, in order to guarantee assembly quality, it's suggested you can update design, move vias away from SMD pads and attach the new gerber files by email.
> Looking forward to your reply, thanks."

[image]

Thank You!

voroskoi, 2025-01-22:
Nevermind, I have used KiCad for uploading the full project and they have accepted it.

closed 2025-01-22 by voroskoi